heidelberg-makerspace/dai-makerspace-do-something #87: Wiki translations

LukasFreeze, 2016-10-24:
@emka installed the translate extension and we are starting to translate policy pages as the rest is too much work for now - here are the bugs we're finding in the process:

* "Kategorie:" namespace does not exist yet - Kategorie:Raumordnung is therefore not an actual translation of Category:Policies

LukasFreeze, 2016-10-25:
I bookmarked "TranslateWiki" a while ago which has notes on translating for MediaWiki - there is a bit on [translating namespace names](https://translatewiki.net/wiki/Translating:MediaWiki#Translating_namespace_names) as well but I am not it will help us as it seems outdated

emka, 2016-10-25:
The page you linked is for another extension, but seems to be outdated.

In the Translate extension's [content translation manual](https://www.mediawiki.org/wiki/Help:Extension:Translate/Page_translation_administration#Markup_examples) they recommend linking to [[Category:Policies/de]] for German translation. I tried that out on the [Issue Tracker](https://wiki.heidelberg-makerspace.de/wiki/Issue_Tracker/de) page.

LukasFreeze, 2016-10-26:
Seems to work! I created a redirect from "Raumordnung" to "Category:Policies/de" as well.

LukasFreeze, 2016-10-31:
Do you know if there is a way to translate our custom sidebar links as well when the site is viewed in German? Would of course only make sense if we translate the targets as well, but right now they really stand out... Maybe we can set new translated parameters rather than typing in the link text directly?

LukasFreeze, 2016-10-31:
also one more note on the linking from categories: some pages are listed twice, e.g. "Help:Events" and "Help:Events/en" on the [policies page](https://wiki.heidelberg-makerspace.de/wiki/Category:Policies), hopefully we can find the reason for that and fix it. "Experts" is listed three times but only because it is not translated until the end where the category tag lies.

emka, 2016-11-01:
Exactly, as you found out the ```[[Category:Something{{#translation:}}``` must be outside the ```<translate> </translate>``` tags.

For the sidebar, I followed the [localised sidebar tutorial](https://www.mediawiki.org/wiki/Help:Extension:Translate/Unstructured_element_translation#Localised_sidebar_tutorial) and it works.

Check out [MediaWiki:Sidebar](https://wiki.heidelberg-makerspace.de/wiki/MediaWiki:Sidebar), [MediaWiki:Sidebar-messages](https://wiki.heidelberg-makerspace.de/wiki/MediaWiki:Sidebar-messages) and [Special:Translate/wiki-sidebar](https://wiki.heidelberg-makerspace.de/wiki/Special:Translate/wiki-sidebar) for modifications.

closed 2016-11-01 by emka

emka, 2016-11-01:
Note that "[the link target cannot be made dependent on the interface language set in the preferences](https://www.mediawiki.org/wiki/Manual:Interface/Sidebar#Links)"...