heidelberg-makerspace/dai-makerspace-do-something #88: Freifunk router can not handle load

emka, 2016-10-26:
The routers we have are quite weak, clients are enough to slow down access for all.

We have three options:
- [ ] add another router - on 5Ghz only - for the ```makerspace``` ESSID (Freifunk is 2.4Ghz only)
- [ ] add more routers (DAI-02 and -03) with **cable uplink** (needs Ethernet cables)
- [ ] use a more powerful router, e.g. TP-Link 1043

emka, 2016-10-27:
The best option for a 5Ghz router I found is the [Asus RT-N66U](https://www.amazon.de/Asus-RT-N66U-Diamond-Dual-Band-Gigabit/dp/B005T3TIBK) (99 EUR), recommended for up to 15 clients. We could also attach a USB hard drive and store some files there.

[TP-Link 1043](https://www.amazon.de/TP-Link-TL-WR1043ND-V3-0-Anschluss-Glasfasermodem/dp/B00NIZXWTU) is 50 EUR, used for ~26 EUR.

15-20m CAT6 cable should be 15 EUR or less.

emka, 2016-11-09:
I started collecting donations specifically to buy the Asus router as soon as we have collected enough funds.

For the ethernet cable I measured
* ~ 2.80m from floor to ceiling
* ~ 14m from wall to window
* ~ 2.5m from the router spot to the side wall

In total, that means ~20-25m cable.

emka, 2016-12-05:
I bought a used RT-N66U, which is not fully funded yet.

emka, 2017-01-26:
We have a 30m cable now, which still needs to be installed.

LukasFreeze, 2017-02-23:
@maikroeder donated a stronger TPLink router and "ethernet via power outlets" adapters which he installed by the audio corner last night. Last thing I heard was it seemed to connect correctly, I'm not familiar with network diagnosis tools myself though so someone might want to double check ;-)

HDValentin, 2017-04-19:
The Router from @maikroeder has an issue. After some minutes it is restarting. All LEDs are on. I put it on the Hack Rack. Perhaps there is another use for this router beside Freifunk.

Have your seen, that there is an offer for an offloader in the Freifunk Forum?
https://forum.ffrn.de/t/futro-s550-offloader-gespendet-wer-hat-interesse/1827/29