heidelberg-makerspace/dai-makerspace-do-something #91: Flash new TinyG PCB

emka, 2016-11-09:
Yannick populated a second SC-TinyG Controller PCB, which still needs flashing.

Atmel Studio required a newer firmware version on our PDI programmer (issue #90). This needs to be resolved before we can continue.

emka, 2016-11-09:
Flashing is done using Atmel Studio, fuses still need to be set.

YannickDupps, 2016-11-09:
I flashed the current master of TinyG (tinyg-master-440.20) 
Fuses are set according to the picture.
[image]

emka, 2016-11-09:
This version is able to talk to ChiliPeppr (now we are one step further than with the first TinyG PCB), but we could not move the axis. Probably the upstream version needs some modifications.

emka, 2016-11-09:
The last firmware from the forum is [this one](https://www.stepcraft-systems.com/media/kunena/attachments/1384/TinyG_2014-11-21.zip).

emka, 2016-11-09:
No luck so far, lets learn more about ChiliPeppr.

emka, 2016-11-10:
The version numbers can be found in ```tinyg.h```, for the firmware from the forum it is
```
#define TINYG_FIRMWARE_BUILD        380.08
#define TINYG_FIRMWARE_VERSION      0.96
```

I think that should be commit ```d8c700f5cca6d0d33aa0d9c24474d098d968e823``` in the upstream project. I checked out that commit and overwrote the files with those from the forum zip file.

```git diff -w``` shows quite a few changes, I still need to have a closer look what has been changed.

emka, 2016-11-20:
The flashing part is done and I think we should try that older firmware first, lets continue in issue #97.

closed 2016-11-20 by emka