heidelberg-makerspace/dai-makerspace-do-something #92: improve laser room

LukasFreeze, 2016-11-10:
Here are a few things we can probably all agree on:
- the space under the laser cutter is a mess
- having to lift your laptop when you arrange your workpiece or perform focusing is pretty annoying
- we would be screwed if we lost the focusing triangle...

So let's tackle these! @Pakue95 suggested mounting frequent-use items such as the log book and focusing triangle to the wall (next to the exhaust system remote) and I think that would be very handy! How about we start with that (and #65 of course!) as the easier steps, and attempt to build some kind of folding laptop table (if it fits) afterwards?

A "caddy" board with holders for the frequently used laser cutter accessories would be nicer than tons of single hooks in the wall, let's collect stuff that should be on it:
- focusing triangle
- hex key (or set?) for lens removal
- cleaning supplies (Q-tips & isopropanol bottle)
- log book/pen
- ...?

The other stuff currently stored under the cutter may have a better place to go or could at least be tidied up a bit (manual, aluminium as weight, marksolid spray, etc)

LukasFreeze, 2016-11-27:
* moved the small items to the top of the machine to stay on this
* @Pakue95 lasered an MDF log book & pen holder
* @emka 3D printed a holder for the focusing triangle and alignment target (that parts needs to be deepened, e.g. by Dremel or reprint)
* I'll cut and mount a board, hopefully already on Tuesday

LukasFreeze, 2016-12-01:
The board and first two holders are up!

The cleaning supplies, suction cups and marking spray should follow soon

mpgeisler, 2016-12-17:
I made an estimate for the dimentions of a laptop mount (based on my ThinkPad + a large safety margin) and made a cardboard cut of it. I put it next to the fire extinguisher in the laser room for now and labled it accordingly. In order for it to not occupy too much space once the mount is made (of wood?) and mounted, we should make it foldable.

LukasFreeze, 2016-12-21:
@mpgeisler I agree, this is what I have in mind (will make a better sketch or 3D model when I have time):

[image]

I'd suggest to rest the front of the laptop holding board on one or two narrow "sticks" rather than another full-sized board.

LukasFreeze, 2017-01-05:
Paul has pretty much the biggest laptop we should/could try to accomodate, and it just about fits on the cardboard template. Unfortunately I forgot to measure it...

LukasFreeze, 2017-01-14:
I made the laptop table today and will tackle the wall part asap, will be documented under "Project:Laptop Wall Mount" in the wiki soon :-)

LukasFreeze, 2017-02-01:
The laptop mount _hangs_ on the wall now but is not yet lockable in the "table" position.

LukasFreeze, 2017-02-16:
Laptop mount is done and working, further improvements are of course encouraged :-)