heidelberg-makerspace/dai-makerspace-do-something #93: Wiki disclaimer

LukasFreeze, 2016-11-14:
A "general disclaimer"/"Haftungsausschluss" is linked at the bottom of any page by default, this page does not exist but the link is not colored red for some reason...we should look into this and fix it as soon as we can to prevent _most_ legal issues.

Maybe [Wikipedia's disclaimer](https://en.wikipedia.org/wiki/Wikipedia:General_disclaimer) could serve as a guide? There are also tons of examples provided by law firms such as [this one](http://www.mustervorlage.net/disclaimer-muster).

LukasFreeze, 2016-11-15:
The same is true for [Project:Privacy policy](https://wiki.heidelberg-makerspace.de/wiki/Project:Privacy_policy).

emka, 2016-11-16:
Please do not write [too much](http://jendryschik.de/weblog/2001/07/06/warum-disclaimer-dem-www-schaden/) in the disclaimer, I think the most important thing is a contact person and address, which we currently provide in [Category:Facilities](https://wiki.heidelberg-makerspace.de/wiki/Category:Facilities) and [Organization](https://wiki.heidelberg-makerspace.de/wiki/Organization).

LukasFreeze, 2016-12-21:
I wrote [a few lines](https://wiki.heidelberg-makerspace.de/wiki/Project:General_disclaimer), how about that? Lawyers who really try will find a hole in anything so I agree with keeping it short

LukasFreeze, 2016-12-21:
Also wrote down my idea of a privacy policy, maybe there should also be something about the sysop's access and how it will not be abused?

closed 2017-01-05 by LukasFreeze