heidelberg-makerspace/dai-makerspace-do-something #95: Improve machining vise

janikmdietz, 2016-11-16:
There are several problems with the [machining vise](https://wiki.heidelberg-makerspace.de/wiki/Vises#Machine_Vises); the quick release is very annoying and prevents you sometimes from unclamping your part. As well the screws for clamping down the vise to the milling/drilling table are way too small. Therefore the following actions should be considered:

- [ ] New screws with bigger diameter (M8 or M10 instead of the existing M5 ones) for clamping down the vise to the milling table
- [ ] New T-slot nuts for clamping down the machining vise to the milling table (size corresponding to the screws mentioned above
- [ ] Removing the quick release of the machining vise due to the earlier mentioned issues

LukasFreeze, 2016-11-17:
Besides the problems with the small machine vise pointed out by @janikmdietz, the large one needs a part (ideally milled metal - once the small one is stable enough) to hold the screw in the frame. Turn it over and you'll see the hole where it needs to go...
As for the screws/t-slot nuts: I measured the milling table a while ago, the grooves are a little less than 10 mm so M10 screws won't fit. Would be interesting to find out whether M8 nuts that fit/can be fitted are available or whether we'd have to make our own! _The slots on the milling table are 18.8 mm wide for the bottom 10.5 mm, then 9.8 mm_

JohannesGruhler, 2016-12-26:
The big machine vise is full working and is secured with M8 screws.