heidelberg-makerspace/dai-makerspace-do-something #97: Find proper Chilipeppr setup for Stepcraft

emka, 2016-11-20:
We want to use [Chilipeppr](http://chilipeppr.com/tinyg) with our TinyG PCB for the Stepcraft.

Last time we tried, the motors did not move yet. We are probably missing some initial settings. I had created these [for grbl](https://wiki.heidelberg-makerspace.de/wiki/Project:TinyG_for_Stepcraft#Configuration_2) in the past, now we need to find out how to set these for TInyG from Chilipeppr.

For homing, I [found](https://github.com/synthetos/TinyG/wiki/Homing-and-Limits-Description-and-Operation) that just sending ```G28``` will not work, but ```G28.2 X0 Y0 Z0``` should be what we need.

emka, 2016-11-20:
There is a wiki page listing all [configuration options](https://github.com/synthetos/TinyG/wiki/TinyG-Configuration-for-Firmware-Version-0.96).

YannickDupps, 2016-11-23:
Chilipeppr can now talk to the Stepcraft. But even though basic communication is working, unfortunately the pinout is wrong. At the moment only the spindle motor can be controlled (with basic on/off commands).

We are going to try three things (in that order) to fix this:

1. Reflash the custom firmware from the old forum thread [https://www.stepcraft-systems.com/forum/zubehoer/427-tinyg-adapter?start=80#10868](url)

2. Try to figure out if it can be fixed via Chilipeppr settings.

3. Compile our own TinyG firmware with custom settings (last resort).

emka, 2016-11-23:
I did step 1 today. Connection via serial console (baudrate 57600) works, but I can not connect in ChiliPeppr on Ubuntu.

These are the settings I read out:
```
[fb]  firmware build            380.08
[fv]  firmware version            0.96
[hv]  hardware version            8.00
[id]  TinyG ID                    3X5332-SYN
[ja]  junction acceleration  100000 mm
[ct]  chordal tolerance           0.001 mm
[st]  switch type                 0 [0=NO,1=NC]
[mt]  motor disble timeout       60 Sec
[ej]  enable json mode            0 [0=text,1=JSON]
[jv]  json verbosity              4 [0=silent,1=footer,2=messages,3=configs,4=linenum,5=verbose]
[tv]  text verbosity              1 [0=silent,1=verbose]
[qv]  queue report verbosity      0 [0=off,1=filtered,2=verbose]
[sv]  status report verbosity     1 [0=off,1=filtered,2=verbose]
[si]  status interval           250 ms
[ic]  ignore CR or LF on RX       0 [0=off,1=CR,2=LF]
[ec]  expand LF to CRLF on TX     0 [0=off,1=on]
[ee]  enable echo                 0 [0=off,1=on]
[ex]  enable flow control         1 [0=off,1=XON/XOFF, 2=RTS/CTS]
[baud] USB baud rate              5 [1=9600,2=19200,3=38400,4=57600,5=115200,6=230400]
[gpl] default gcode plane         0 [0=G17,1=G18,2=G19]
[gun] default gcode units mode    1 [0=G20,1=G21]
[gco] default gcode coord system  1 [1-6 (G54-G59)]
[gpa] default gcode path control  2 [0=G61,1=G61.1,2=G64]
[gdi] default gcode distance mode 0 [0=G90,1=G91]
[1ma] m1 map to axis              0 [0=X,1=Y,2=Z...]
[1sa] m1 step angle               1.800 deg
[1tr] m1 travel per revolution    1.250 mm
[1mi] m1 microsteps               8 [1,2,4,8]
[1po] m1 polarity                 0 [0=normal,1=reverse]
[1pm] m1 power management         0 [0=remain powered,1=shut off when idle]
[2ma] m2 map to axis              1 [0=X,1=Y,2=Z...]
[2sa] m2 step angle               1.800 deg
[2tr] m2 travel per revolution    1.250 mm
[2mi] m2 microsteps               8 [1,2,4,8]
[2po] m2 polarity                 0 [0=normal,1=reverse]
[2pm] m2 power management         0 [0=remain powered,1=shut off when idle]
[3ma] m3 map to axis              2 [0=X,1=Y,2=Z...]
[3sa] m3 step angle               1.800 deg
[3tr] m3 travel per revolution    1.250 mm
[3mi] m3 microsteps               8 [1,2,4,8]
[3po] m3 polarity                 0 [0=normal,1=reverse]
[3pm] m3 power management         0 [0=remain powered,1=shut off when idle]
[4ma] m4 map to axis              3 [0=X,1=Y,2=Z...]
[4sa] m4 step angle               1.800 deg
[4tr] m4 travel per revolution  360.000 mm
[4mi] m4 microsteps               8 [1,2,4,8]
[4po] m4 polarity                 0 [0=normal,1=reverse]
[4pm] m4 power management         1 [0=remain powered,1=shut off when idle]
[xam] x axis mode                 1 [standard]
[xvm] x velocity maximum        600.000 mm/min
[xfr] x feedrate maximum        600.000 mm/min
[xtm] x travel maximum          150.000 mm
[xjm] x jerk maximum       20000000 mm/min^3
[xjh] x jerk homing        20000000 mm/min^3
[xjd] x junction deviation        0.0500 mm (larger is faster)
[xsn] x switch min                1 [0=off,1=homing,2=limit,3=limit+homing]
[xsx] x switch max                0 [0=off,1=homing,2=limit,3=limit+homing]
[xsv] x search velocity         500.000 mm/min
[xlv] x latch velocity          100.000 mm/min
[xlb] x latch backoff             2.000 mm
[xzb] x zero backoff              1.000 mm
[yam] y axis mode                 1 [standard]
[yvm] y velocity maximum        600.000 mm/min
[yfr] y feedrate maximum        600.000 mm/min
[ytm] y travel maximum          150.000 mm
[yjm] y jerk maximum       20000000 mm/min^3
[... 89 lines cut ...]
[g59x] g59 x offset               0.000 mm
[g59y] g59 y offset               0.000 mm
[g59z] g59 z offset               0.000 mm
[g59a] g59 a offset               0.000 deg
[g59b] g59 b offset               0.000 deg
[g59c] g59 c offset               0.000 deg
[g92x] g92 x offset               0.000 mm
[g92y] g92 y offset               0.000 mm
[g92z] g92 z offset               0.000 mm
[g92a] g92 a offset               0.000 deg
[g92b] g92 b offset               0.000 deg
[g92c] g92 c offset               0.000 deg
[g28x] g28 x location             0.000 mm
[g28y] g28 y location             0.000 mm
[g28z] g28 z location             0.000 mm
[g28a] g28 a location             0.000 deg
[g28b] g28 b location             0.000 deg
[g28c] g28 c location             0.000 deg
[g30x] g30 x location             0.000 mm
[g30y] g30 y location             0.000 mm
[g30z] g30 z location             0.000 mm
[g30a] g30 a location             0.000 deg
[g30b] g30 b location             0.000 deg
[g30c] g30 c location             0.000 deg
```

emka, 2016-11-23:
I sent ```G28.2 X0 Y0 Z0``` in the serial console, the result was changing positions in the serial console but no motor moving.

emka, 2016-12-13:
Closing as this is only related to our TinyG board, not the grbl approach.

closed 2016-12-13 by emka