heidelberg-makerspace/dai-makerspace-do-something #98: radiator in the main room can't be regulated

LukasFreeze, 2016-11-22:
The radiator (Heizkörper) by the Mac/to the right of the main room windows is always blasting away and makes the Space too hot as well as wasting a ton of energy.

It may be fixable by reattaching the thermostat a different way, does anyone have experience with that?

YannickDupps, 2016-11-23:
I can take a look at it the next time I'm around. It's usually just a question of reattaching properly and maybe adding some WD40.

LukasFreeze, 2016-11-24:
I couldn't stand it any longer, it was a different kind than I know from home but after doing some quick [YouTube research](https://m.youtube.com/watch?v=AfSgma0cEVE) I wasable to fix it.

Sorry @YannickDupps - I'll let you have the next thing ;-)

closed 2016-11-24 by LukasFreeze