heidelberg-makerspace/dai-makerspace-do-something #99: milling table fine tuning / important to know

JohannesGruhler, 2016-11-23:
Problems:                                                            
- the milling table is not 100% in the right angle.
- the Z-Achse can move down because of the Vibration from the motor.  (ask either [Lukas](https://wiki.heidelberg-makerspace.de/wiki/User:Lukas) or [Johannes](https://wiki.heidelberg-makerspace.de/wiki/User:Johannes) vor more Information’s)

wiki page: [milling table](https://wiki.heidelberg-makerspace.de/wiki/Milling_Table)

LukasFreeze, 2017-01-13:
@JohannesGruhler apparently found out that the screws on both sides of the Z sled (three slotted screws each) may need to be tighter. 

It seems very difficult to turn them, maybe the part surrounding them is countered against them and we need to make a special tool, perhaps out of an old screwdriver or bit? Look at it and you'll see what I mean, otherwise I'll try to sketch it ;-)

LukasFreeze, 2017-01-27:
Clemens confirmed that the outer "screws" are most likely counter nuts and suggested using the term "Schwalbenschwanzführung" for some research. Some play in the Z axis can not be avoided due to the nature of a threaded rod as driving element but we should be able to tighten it enough so the vibrations don't cause problems anymore.
We'll also need to lubricate the part properly and more regularly, not just WD40!

LukasFreeze, 2017-02-20:
this issue should be referenced from a label on the machine so users find out about it before working.