heidelberg-makerspace/dai-makerspace-do-something #103: Prusa B Z-offset or bed calibration is off again

ragflyer, 2016-12-12:
In the front left corner it prints into air while in the back right corner the nozzle touches the bed.

Also the flow does not seem to work as in Prusa A, sometimes it does not print anything, sometimes it flows into areas that should be empty. I tried different filament and temperatures today which did not improve the behaviour.

LukasFreeze, 2016-12-13:
we should check whether the bed is bent, it's weird that it doesn't work so shortly after the last calibration!

beanobeano, 2016-12-13:
I'm at the makerspace now and had a quick look. 
The cable for the Y-Stepper was looped under the back left corner of the acrylic frame. 
This might of caused the printer to twist slightly and possibly caused the issue.
I'll tie up the cable so it can't slip under the frame again.

beanobeano, 2016-12-13:
... and both nuts at the back on the M8 threaded rods were loose.

LukasFreeze, 2016-12-13:
phew, that's bad! Good thing you noticed, otherwise we would go mad over calibrating over and over again... I think André mentioned he mounted his i3 to a wooden board for transport, maybe we should find a good way to do that and then calibrate?

beanobeano, 2016-12-13:
I was just thinking a similar thing, as there was a lot of crap on the table under the printer including some partial prints and some filament clippings from changing filament. Any of these under one of the corners could have also caused it to twist (especially with the loose nuts). 
Fixing the printer to a stable flat board would address this permanently or relatively permanently :-) or at least improve the situation.
I'll check for warping of the bed too... just in case that is also an issue. Hopefully not.

beanobeano, 2016-12-13:
Bad news. There is quite a substantial warping across the bed in the X direction. Raised in the middle compared to the left and right sides - by over 1mm. It's not so bad in the Y direction.

LukasFreeze, 2016-12-13:
Do you think loosening a screw or two might help? I think we had a similar problem with Prusa A once and ended up exchanging the springs between the bed and its carriage for weaker ones. Otherwise I guess we'd have to bend it back _carefully_ :-P

beanobeano, 2016-12-13:
It's quite warped. We can try loosen the screws but I think it's already permanently warped.

Andi-Heidelberg, 2016-12-13:
I dismounted the bed, and it is warped in x and y-direction. Sorry for the bad news...

Pakue95, 2016-12-15:
I bend the bed back as good as I could. Also the printer is now secured on a piece of wood with some brackets which should prevent further miscalibration in the future. 

**To Do**

1. mount the bed again
  * otherwise the Z sensor won't work
2. Recalibrate the printer

Ianmcmill, 2016-12-15:
Hello!
I visited your makerspace yesterday as a guest and looked around a bit. 
I also have a very slightly bent bed. 2mm [MK3 ALU-Heatbed Dual Power](http://www.reprap.me/alu-heatbed-mk3.html). I directly print onto it's surface with blue painters tape.
I got the bed for 15€ with minor scratches from ebay.
I am using cheap and low tension springs from my local hardware shop. They work nicely.

If you already use an (permanently attached) inductive probe like the  you could enable ABL (auto bed leveling) in Marlin/Repetierfirmware (as of Marlin 1.1.0 RC7 quite stable) and still print on a bent bed without hassles.
There are several bed leveling methods of which the bilinear method should work best with an uneven i.e. bend bed
` * - BILINEAR`
 `*   Probe several points in a grid.`
 `*   You specify the rectangle and the density of sample points.`
 `*   The result is a mesh, best for large or uneven beds.`

LukasFreeze, 2016-12-15:
We do have bed leveling, but I think the firmware just calculates a plane and with a bent bed parts of that virtual plane end up under or too far over the actual bed. Not sure if this "bilinear" method is included in our firmware, but if it works with bent beds it's yet another reason for an upgrade, thanks!

To whoever will mount the bed: please make it as square as possible using the leadscrews (slide x and y with the power off to check) and maybe replace the springs with softer ones (found in the workshop material cabinets, but maybe not four of a kind) to reduce further stress on the plate :-)

ragflyer, 2017-02-04:
Bed is leveled and taped now, I adjusted the firmware on the slicing laptop and did some test prints in each corner and center. Could somebody with more experience check those prints and let me know if I should put the firmware on github?
It looks like the bed is marginally higher in the center but current setting seems to work in every location on the bed.

closed 2017-02-04 by ragflyer

LukasFreeze, 2017-02-04:
A while ago I set the first layer thickness to 0.3 mm regardless of any
other settings, so it might be okay if it is just slightly higher in the
middle - would be interested to see your prints sometime!

Andi-Heidelberg, 2017-02-04:
Great job! Thumbs up!

LukasFreeze, 2017-02-06:
looking at @ragflyer 's test prints I would say the nozzle gap is good on the left, slightly too small in the center and slightly too big on the right. This may not be fixable without replacing the aluminum plate, but it's no problem for printing! Just print in the center for a very flat bottom and on the right if you're afraid your print won't come off ;-)
I printed a large piece across the majority of the bed area and it looks perfect!