heidelberg-makerspace/dai-makerspace-do-something #104: Create Laser Cutter Guide

beanobeano, 2016-12-13:
I think it might be useful to have a guide for positioning pieces of material at a defined offset from the automatic zeroing point of the Laser Cutter on startup.

For example: If you create a workpiece at some point and then later want to modify it to add a couple of extra mounting holes or to cut out a bigger window for an upgraded display, etc. You could use the guide to help position your piece on the bed to make the modification.

beanobeano, 2016-12-13:
I designed and cut an L shaped Acrylic piece that when placed on the bed will allow positioning of the workpiece with corner at 10cm,10cm offset from the auto zero point.

I hope this helps. We just need to find a nice place to put this.

beanobeano, 2016-12-13:
Thanks Patrick for the support in cutting this.

LukasFreeze, 2016-12-15:
I would suggest storing this and other jigs under the machine once #92 is done (too messy right now)