heidelberg-makerspace/dai-makerspace-do-something #106: Find good CAM solution

emka, 2016-12-14:
We need a good CAM solution, a few options are [listed in the wiki](https://wiki.heidelberg-makerspace.de/wiki/Stepcraft#CAM_options).

The ideal solution should
* run on all major platforms (Windows, Mac, Linux)
* calculate paths fast enough
* read different input formats (SVG, DXF, STL)
* be free (as in beer) and ideally be free software (as in freedom)

emka, 2016-12-17:
BlenderCAM is very promising, still need to figure out the details

LukasFreeze, 2016-12-18:
Onshape suggests [Kiri:Moto](https://appstore.onshape.com/apps/CAM/EAAEWYIOMQKBENEMYW2N7MF253CT4WYL6SUJGEY=/description?utm_campaign=Onboarding&utm_source=hs_automation&utm_medium=email&utm_content=33221160&_hsenc=p2ANqtz-8oU7KcQvM5bxdzT8Lk0o8XG2HckpWfr5TOYglDviDHTUt8azH8LPcVigL6JKNgo5PpiQeosVr3PpKUs4rIuGCGpbIZ-wQflzXJTj-_3khK-CPg5a8&_hsmi=33221160) (free, but cloud-based), have you heard of that?

emka, 2016-12-18:
Yes, Kiri:Moto is in the list as well. When I tried launching it from OnShape, my part did not import because of an error.

emka, 2016-12-19:
> BlenderCAM looks nice, but to design the things you must learn 3D modeling with Blender :-/

Actually you can import STL or SVG designed in another program. Once we figured out which default settings we need, they could be realized by a standalone Python script without even opening the GUI.

Kiri:Moto seems to be quite beginner friendly, it could be the best solution to get quick results.

The screenshots in the above article show AXIS from LinuxCNC to transfer G-code. We will not have a standalone computer for LinuxCNC, but the LinuxCNC manual is a great resource anyway. I listed [other options](https://wiki.heidelberg-makerspace.de/wiki/Stepcraft#Sending_G-code_to_grbl) in our wiki.

For CAM, the article mentions FreeCAD and HeeksCAD / HeeksCNC. FreeCAD path generation is quite new, @Pakue95 and me were not able to use it. HeeksCAD looks promising if someone is willing to find out more about it.

I have tried Camotics for simulation, it worked well. You need to define your tools in there again, simulation can be paused or accelerated, but not shifted in time.

emka, 2016-12-30:
Kiri:Moto does not keep my saved settings (quite annoying to adjust everything each time) and does not support SVG. Could be workarounded by extruding SVGs, if someone is patient enough.

emka, 2016-12-30:
The combination of [Millcrum](http://millcrum.com/) and [cnc](https://github.com/cheton/cnc) worked great. Millcrum is 2D only.

emka, 2016-12-30:
BlenderCAM works great, too.

closed 2016-12-30 by emka

emka, 2017-01-12:
We got best results with Estlcam (proprietary, but affordable) so far. Let's see how well Fusion 360 works for us (CAM on imported STL did not work immediately for me).

LukasFreeze, 2017-01-13:
I've come across ["vCarve Makerspace Edition"](http://www.vectric.com/products/makerspaceedition/how-does-it-work.html) which may also be interesting in the future, it allows for unlimited "clients" to prepare everything at home and a licensed version to have in the Space ($99 a year) which controls the machine(s).