heidelberg-makerspace/dai-makerspace-do-something #107: Mount electronics to wall

emka, 2016-12-14:
Put electronics in a nice case (e.g. laser-cut acrylic) and screw it to the wooden wall. Keep in mind where the cables will go.

MichelDC, 2017-07-20:
Not the final / best / nicest solution, but a bit cleaner and the milling dust does not get onto the electronics anymore. I have not checked if the Arduino actually gets warm. If yes, I would need to add some ventilation slits.

[image]
[image]

emka, 2017-07-20:
Looks great and much cleaner!

Regarding the air flow: Depending on the feed rate, the stepper motor drivers could run hot. If you are experiencing lost steps, resulting in incorrect movements, that will probably be the cause. There are also no heatsinks on the drivers yet.