heidelberg-makerspace/dai-makerspace-do-something #108: Build spring holder for tubes

emka, 2016-12-14:
The Stepcraft vacuum hose and sealing air tube are moving with the X and Y axis and should be held above the table by a spring holder.

LukasFreeze, 2016-12-21:
In case finding a proper helical spring turns out too hard, we could also try retractable keychains (should have one at home) or something like a bunch of filament scraps from the ultimaker taped into a flexible stick

HerrBaumhaus, 2017-02-13:
I don't get what is actually needed here. I have some spare springs, what exactly are they for? Dimensions etc.?

LukasFreeze, 2017-02-13:
They should be large but soft, holding the vacuum hose/pneumatic hose/power
line up and out of the tool path.

We thought of springs as they should allow tool movement, a spring-loaded
string (as in some keychains/dog leashes) could also be an option