heidelberg-makerspace/dai-makerspace-do-something #109: enable Special:EmailUser in the wiki

LukasFreeze, 2016-12-15:
It would be nice if members could eMail each other quickly through the wiki - fewer addresses would need to be out in the open and potentially spammed but communication, especially in relation to expertise or calendar entries, would be easier.

There is a special page that, when enabled, shows up in the toolbox on the left when visiting a user page with a confirmed eMail address. Access to this feature, like writing, can be blocked for everyone except certain groups (I suggest only _members_ can use it).

Here's what I found on the topic:
* [Wikipedia's guide on using the feature](https://en.wikipedia.org/wiki/Wikipedia:Emailing_users)
* [MediaWiki page about blocking users](https://www.mediawiki.org/wiki/Help:Blocking_users)
* [list of MediaWiki permissions including blockemail](https://www.mediawiki.org/wiki/Manual:User_rights#List_of_permissions)

emka, 2016-12-15:
I wanted to enable this feature, but I do not see how to restrict it to a certain group (if it was possible, I would prefer _write_ to _members_). The ```blockemail``` permission (usually sysop only) allows to manually block other accounts from sending email.

As enabling it would mean that any newly created spam account could send email to other accounts from our server, I will keep it disabled until sending permission can be restricted to a group.

closed 2016-12-15 by emka

LukasFreeze, 2016-12-15:
It's definitely possible, mediawiki and wikipedia use it that way - does [this rights list](https://www.mediawiki.org/wiki/Special:ListGroupRights) help? There is a right called "sendemail" that is granted only to the group "Users" in their case.

emka, 2016-12-15:
Thanks, that is the permission I was looking for. Enabled it now for users in the ```Write``` group.

LukasFreeze, 2016-12-15:
Awesome! :-D