heidelberg-makerspace/dai-makerspace-do-something #111: Gmail blocks some mailman messages

LukasFreeze, 2016-12-18:
With more users in our various mailing lists fewer and fewer messages are falsely classified as "spam", but a major problem persists: Gmail blocks some eMails that it considers suspicious entirely and the addressees can not see them or teach it that it was not in fact spam.

It's likely that I have provoked this when switching all previous newsletter subscribers to the new list, but adhering to Gmail's [bulk sender guidelines](https://support.google.com/mail/answer/81126?hl=en) more closely might solve this issue.

LukasFreeze, 2016-12-18:
It starts with "authentication & identification": as far as I know, we do have consistent "from:" addresses but several IPs from which Mailman will send. The third advice is to "Keep valid reverse DNS records for the IP address(es) from which you send mail, pointing to your domain" which is not entirely clear to me yet.

emka, 2016-12-18:
We have correct reverse DNS entries for [IPv4](http://mxtoolbox.com/SuperTool.aspx?action=ptr%3a78.47.161.102&run=toolpage) and [IPv6](http://mxtoolbox.com/SuperTool.aspx?action=ptr%3a2a01%3a4f8%3ac17%3a1e51%3a%3a2&run=toolpage).

What we do not have is SPF and DMARC records, if anyone knows these well enough please contact me.

emka, 2016-12-18:
I think our SPF record should look like this:
```
v=spf1 ip4:78.47.161.102 ip6:2a01:4f8:c17:1e51::2 -all
```
Can anyone confirm?

LukasFreeze, 2016-12-18:
I don't have any experience of course, but from the [Google support page](https://support.google.com/a/answer/178723) I understand it should rather be 

` v=spf1 include:_spf.google.com ip4:78.47.161.102 ip6:2a01:4f8:c17:1e51::2 ~all`

They say "-all" may cause delivery problems.

emka, 2016-12-18:
```_spf.google.com``` is for sites hosted on Google servers. I added
```
v=spf1 ip4:78.47.161.102 ip6:2a01:4f8:c17:1e51::2 ~all
```
please let me know if anything fails.

emka, 2016-12-18:
We are also missing valid DKIM signatures.