heidelberg-makerspace/dai-makerspace-do-something #114: Stepcraft control computer

emka, 2016-12-30:
I am very happy with the [cnc](https://github.com/cheton/cnc) interface to control our Stepcraft. There are a few problems:
* Everyone needs to install and configure it.
* The room is quite cold, not warm enough to stay there in winter.
* You can not remove your laptop during the milling process.
* It would be great if you could watch from outside.

To address these problems, I suggest [installing the interface on a Raspberry Pi 3](https://github.com/cheton/cnc/wiki/Raspberry-Pi-Setup-Guide) and make it accessible via its own wifi. It comes with webcam support, adding a webcam should be easy as soon as we have one.

Then running a job should be easy: join the wifi network, submit the job and disconnect whenever you like.

emka, 2016-12-30:
The emergency stop (#115) should work before we do that.

emka, 2017-01-26:
Raspberry Pi 3 arrived.

emka, 2017-06-08:
I installed cncjs on it quite a long time ago but never got around actually using it.
@MichelDC, could you give an update on the current status?

An alternative to cncjs could be [Laserweb 4](http://cncpro.co/).

MichelDC, 2017-06-08:
I only used cncjs on the Raspberry Pi once and then it worked. I normally used the app on my Mac, as there's no static IP for the Pi.
I have some problems with cncjs, as it seems to cut wrong paths from gcode from Fusion 360 when the two paths are not connected. However, I have not tried if this also happens with cncjs on the Pi, but I realized that chilipeppr seems to simulate the path correctly, so I assume it's a cncjs problem.

MichelDC, 2017-06-14:
I tried Laserweb 4: On the positive side it solves the problem with the wrong paths in cncjs. On the negative side, it's a bit clumsy (compared to cncjs) and the preview window sometimes seems to show nothing, but I guess this might be depending on the OS.
I also did not manage to get it working with an SVG from Inkscape, but I did not investigate further if it is a problem with the SVG or with Laserweb.

MichelDC, 2017-07-06:
Following up on this: The new version of Laserweb 4 improved on some issues, but still has some graphics bugs (and I still don't understand what the preview window _exactly_ shows, but whatever :))
I managed to get it to work with a simple SVG from Inkscape, and one can set all needed parameters, I think.