heidelberg-makerspace/dai-makerspace-do-something #116: laser exhaust system asks for maintenance

LukasFreeze, 2016-12-31:
Our exhaust system (VacuAir UML 500, not listed in [cameo's selection](https://www.cameo-laser.de/de/systeme/absauganlagen/) anymore) has a flashing red/green light and the remote reads "Sichtkontrolle" or "Wartung Akt. Kohle" if you confirm having looked inside.

It appears this is more than just a "I've run for 100 h, please check me" thing, unfortunately there does not seem to be a manual... I'll get in touch with our technician and ask for one.

LukasFreeze, 2017-01-01:
I got the manual as a pdf, if we really don't have a copy in the black binder under the laser I'll print it on Wednesday. Nothing on specific error messages, though...

LukasFreeze, 2017-01-04:
I received the missing document after a nice chat with a cameo guy and know what to do, will print the manual tonight!

LukasFreeze, 2017-01-05:
This remains more difficult than it should, it seems a passcode is required to access the relevant menu entry...I'll see if I can get that as well -.-

Pakue95, 2017-01-06:
The code worked! This unlocks the menu and you can check off "Sichtkontrolle" and other settings. I also checked off "Aktivekohle" as our filter is new and shouldn't need changing this early. Is was probably missed to reset in the factory.

closed 2017-01-06 by Pakue95