heidelberg-makerspace/dai-makerspace-do-something #117: Find and build material storage solution

emka, 2017-01-02:
Several members are storing large acrylic or MDF sheets in the rooms. This is understandable as they are too large to transport them on a bike. However, currently there is no real place to store them. Some are sticking out of member's boxes (with pointy edges), some have overdue parking tickets and others are stored with a name, but no parking ticket.

We need a better solution to allow storing small quantities of oversized sheets.

LukasFreeze, 2017-01-05:
Behind the couch would be one option - @jasperclaudio, any idea what is going to happen with those maps??

jasperclaudio, 2017-01-05:
Together with Jaro I'll finally think of good solution on what to do with them. so expect them to be some time during the next week.

LukasFreeze, 2017-01-22:
They were still there yesterday, can we please get rid of them by February so we can establish the new storage while we are cleaning up after the job fair anyway?

JaroEiermann, 2017-01-22:
Hey yes, we will need them for several workshops in the  makerspace in the future, so the best would be to have a place for them in the makerspace.

jasperclaudio, 2017-01-22:
Amir cleaned up the Hausmeister room and a lot of the stuff from the Sprachschule moved into their new study rooms next door. This means there should be space for them. I'll move them there tomorrow 😉

LukasFreeze, 2017-02-02:
The space behind the couch is free now, there would just need to be a placeholder near the floor so the couch won't crush anything if pushed back. The material left in the laser room were all moved next to the large workbench yesterday, but that is not an actual solution!

**There will be no new parking permits for sheet materials until this issue is resolved.**

LukasFreeze, 2017-02-02:
We could use a part of the workbench's lower surface for smaller scraps and the couch space for larger pieces, either way both need a "cage" to prevent pieces sliding around and as a size limit, otherwise we'll have stuff lying around everywhere again in a matter of weeks...

emka, 2017-02-02:
A scaled down version of [this](http://www.thingiverse.com/thing:26692) with more, but thinner compartments might fit behind the couch and possibly below the shelf near the janitor's office. For smaller or partially cut parts I think flat drawers would be better, it may be possible to build something like that below the large workbench (?).

LukasFreeze, 2017-03-04:
Ed and I fitted the left third of the large workbench with a "fence" behind which sheet materials can be stored. Please put your name on your piece if it's supposed to be private, or a suggested donation if you would like to share!

**A few** bigger pieces can also be stored between the workbench and the exhaust filter. Pieces outside these storage areas require valid parking permits or they are free for whoever wants them!

closed 2017-03-04 by LukasFreeze