heidelberg-makerspace/dai-makerspace-do-something #118: Prusa A needs Z calibration

LukasFreeze, 2017-01-12:
The "Prusa i3 A" printer starts out too low, leading to high pressure and potentially (as last night) clogging in the nozzle. To fix this, change the Z offset - probably needs to be 0.2 mm further from 0 which will open up the gap to the bed - and document the new values along with anything missing from the currently very short wiki page on [Prusa i3 calibration](https://wiki.heidelberg-makerspace.de/wiki/Prusa_i3_Calibration).

Pakue95, 2017-01-13:
Andre noticed on Wednesday that the right linear bearing came loose and is now slipped down from the printed collar. 

We need to reinsert it and made secure it from further slippage in the future.

LukasFreeze, 2017-01-13:
True, that shouldn't be the problem for the Z error but of course we need
to fix it as well. A dab of hot glue worked well in the past, this is the
first time one came off again...

LukasFreeze, 2017-01-19:
Johannes changed the setting last night ([see wiki](https://wiki.heidelberg-makerspace.de/wiki/Prusa_i3_Calibration)), but went a bit too far - a Z offset of -0.6 resulted in the print head being too low, 0 is a bit high and produces an imperfect first layer. Something in between, maybe -0.3 or -0.2, should work and can be fine-tuned through _Advanced -> Initial Layer Line Width_ in Cura.

JohannesGruhler, 2017-01-25:
I  changed the Z offset to -0.1mm
The print result is good

ragflyer, 2017-01-27:
I tested it too, the first layer barely works but I think we should decrease the offset further to -0.2 or -0.3 so the first layer connects properly.

LukasFreeze, 2017-01-27:
That's weird, it looked perfect on Wednesday! Could you show me your print?
And did you slice with the newest Cura settings?

JohannesGruhler, 2017-01-29:
I showed you the newest print already and you said that we can lose the issue.

closed 2017-01-29 by JohannesGruhler

LukasFreeze, 2017-01-29:
But Marius seems to have printed something after that ;-) We'll see

ragflyer, 2017-02-04:
Back left corner was at least 3mm higher, i leveled the bed and tightened the sensor again.
Then I adjusted the firmware (see last changes at top of Configurations.h) and tested in front left corner.
Seems good now but please check other corners before you do a production print.