heidelberg-makerspace/dai-makerspace-do-something #119: Build cyclone dust separator

emka, 2017-01-17:
Build a cyclone dust separator (Zyklonabscheider) for wood dust created during milling. This will avoid to use too many vacuum cleaner bags.

There is a 3D printable design at http://www.thingiverse.com/thing:1736769 which might fit our printers, but will probably print 2x 12-20 hours and use lots of filament. There might be other options.

LukasFreeze, 2017-01-17:
Good idea, but I think I've seen simpler solutions using buckets or so -
I'll see if I can find it again!

We could also try and fit the whole thing to a Systainer-like base that
will fit on top of the Festool shop vac (two green plastic sliders would
hold it iirc).

LukasFreeze, 2017-01-17:
The one I originally had in mind actually used the cyclone from a broken vacuum cleaner, but here are two different approaches:
* [Matthias Wandel's cyclone](http://woodgears.ca/dust_collector/cyclone.html), made from buckets and wood
* [cheap & quick version](https://www.youtube.com/watch?v=9w8jXS8Fjh8)

emka, 2017-02-28:
I have seen bucket based solutions implode because the walls were to weak for the vacuum (or stabilizing inserts were not placed correctly).
[Einhell](http://www.einhell.de/de_de/blog/artikel/zyklonabscheider-selber-bauen.html) has some (German) instructions as well, I would prefer that if we are able to get such a pipe.

LukasFreeze, 2017-02-28:
Those look very nice!

Unfortunately, as far as I remember (and still think after a quick online search), Bauhaus only has much narrower pipes - I'll ask a friend with construction contacts but can't promise anything.

topeban, 2017-03-19:
In Stock :
Short Piece of  Narrow Pipe  Outer Diameter 315 x 10,0 mm   Length min 235 mm

LukasFreeze, 2017-03-23:
I had no success finding a source of wide pipes so far, but someone also building a dust separator donated a massive piece (bit shorter than in the instructions though). It's sitting by the workbench, unfortunately I don't know any other details

topeban, 2017-03-23:
Hello Lukas ,
Yesterday I brought this Pipe to the Makerspace , an tell someone it is for the cyclone dust separator.

I become this pipe from a Construction Company  out of the  Plastic Garbage.
        (And YES I ask before 8-)  
        (2-3 Years ago , Originally I plan to use this pipe as a mould for wood-Concrete Bird-House)
And I have another one too , with the same Dimension.

LukasFreeze, 2017-03-23:
Awesome, thank you! You probably talked to Patrick, I was really busy :-P

LukasFreeze, 2017-08-05:
Whenever we do tackle this build, the ["Thien Baffle"](http://www.jpthien.com/cy.htm) and its many variations may also be good to look at!