heidelberg-makerspace/dai-makerspace-do-something #120: improve workshop lighting

LukasFreeze, 2017-01-23:
The lights on the ceiling don't do much good as you're usually casting a shadow on your work, and the LED ring lamp keeps being moved between the workbenches. The LED ring lamp on the e.Bench can get in the way of the laser room door or the people passing through.

My suggestion: move the e.Bench lamp to the unlit workbench and make something like [this LED strip bench light](http://www.instructables.com/id/DIY-LED-Workbench-Light-From-LED-Strips/) for the e.Bench.

Pakue95, 2017-01-23:
Looks nice!
We could ask Johannes to use one or two of his high power LED chips :D
Another idea could be mounting a aluminium extrusion and use a slider to move the light

emka, 2017-01-23:
The lamp also provides a magnifying glass which has already been used for soldering. The replacement solution should include another magnifying option as well.

LukasFreeze, 2017-01-23:
I wasn't sure how much that was actually used - there are also two "third
hands" with magnifying glasses among the soldering equipment, are those
sufficient?

Pakue95, 2017-01-23:
I use the lamp every time when soldering. 
I think it's mich easier to work with

LukasFreeze, 2017-02-17:
@Pakue95 found a desk lamp in the trash, just needs a nice E27 bulb and hopefully no repair!

LukasFreeze, 2017-03-04:
We installed the lamp on the Green Workbench, looks pretty cool...like it's growing out of it :-D

closed 2017-03-04 by LukasFreeze