heidelberg-makerspace/dai-makerspace-do-something #123: main room lights are crap

LukasFreeze, 2017-02-01:
Two of the main room lights are broken (once again) and a third one fell off the wall, it's time we do something!

Let's collect ideas on better solutions (brighter & light rather than heat would be nice), I'll check whether we'll get support from the DAI or are at least allowed to fix this.

Some facts:
* currently 8 lights, _facing the ceiling_
* must be dimmable

jasperclaudio, 2017-03-06:
I fixed the light next to the sewing rack.
Are there any new ideas on how to improve the light situation? Klaus the old Hausmeister showed me where he stored all the parts necessary to fix the spots on the wall. But I have to admit it's not an ideal system. Especially because the individual parts are quite expensive...

emka, 2017-03-13:
We probably need to replace them by dimmable LEDs, so far I think nobody has researched options yet.