heidelberg-makerspace/dai-makerspace-do-something #125: Prusa B box / air filter

ragflyer, 2017-02-21:
I made a box for Prusa B to gain some experience with materials we don't want to inhale while printing.
Temperature inside is much more stable now -> better inter-layer-adhesion.
Now I would like to add some kind of air filter and would like to know about your opinion.

air circulation
Blowing the air out of the box through a filter would create a vacuum and avoid leaks of bad air, but it would also create a flow that is counter-productive for keeping the temperature inside. Circulating air inside the box through a filter would be better for keeping the temperature but leakage could be higher. In a test print today there was no noticeable leak, but it will increase if we add a maintenance window.

filter
There are lots of filter ideas on thingiverse, but I do not feel competent in that area to decide if these good enough or if we need certified cartridges like those used in face masks.

contoller
To externalize the user interface we need a little longer cables. There are two 10-wire flat cables with 2x5 Pin connectors. Maybe somebody still got some of these at home, otherwise I'll order the correct length.

Anything else I forgot?

emka, 2017-02-21:
I also think circulating inside the box is better. Andre designed a filter for his printer, basically a box filled with carbon filter pellets and a fan moving the air through it. An easy way to replace the filter material without disassembling everything would  be useful.
If you decide to blow air outside, you also need to make sure that the box will not implode.

ragflyer, 2017-02-23:
On Thingiverse they have some boxes held together by rubber bands for easy maintenance.

I would design one for a radial fan as these have a much higher pressure, which we need to push the air through the filter.
On outlet side I thought of a mechanical slider so we can control how much of the output should go out of the box and how much should circulate.

Marco mentioned yesterday the electronics and power supply might get too hot and should be moved outside. For this we need to extend some cables so for the moment I put the thermometer inside and the temp was around 41°C after 1 hour of printing at 245/95°.

LukasFreeze, 2017-02-23:
Things seem to be going very well, I hope your print will also fulfill your
expectations in the end!

Please check the post boxes next to Prusa A when you have the chance, there
may still be a grey radial blower in one of the open ones (unless Janik
kept it) :-)

If you end up buying extra cables, just remember give me the receipt so I
can reimburse you!