heidelberg-makerspace/dai-makerspace-do-something #126: Prusa B heatbed subframe melting

ragflyer, 2017-02-23:
After some ABS prints the subframe bent to the point it scratched the frame. We talked it through yesterday and have several options.

1. (1,39 EUR) get one of these https://www.bauhaus.info/metallprofile-kunststoffprofile/kantoflex-quadrat-u-profil-/p/10522007 and put a piece on each side of the acrylic part to align the edges to the center
2. (≈29 EUR) aluminium subframe Marco mentioned yesterday, he will come back next week with a price for the makerspace
3. (≈80 EUR) get a full steel or aluminium frame to avoid other parts of the frame melting next

Until then we need to watch closely and hold the heatbad cable while homing to avoid it getting stuck.
Printing first two layers on 105° bed temp then going down to 95 or 90° seems to work well.

LukasFreeze, 2017-02-23:
>    kunststoffprofile/kantoflex-quadrat-u-profil-/p/10522007

The full frame would of course be the most stable option, maybe Marco's
friend can also give us an offer for manufacturing that? While 80 € (ebay I
guess?) may be affordable from the 3DP cash right at the moment, it would
be nice to support a local manufacturer!

LukasFreeze, 2017-02-26:
Here's the [Y carriage on "Hazefactory"](https://www.hazefactory.eu/shop/i3-y-aluschlitten/) - unfortunately, the whole frame is not offered. Instead of buying a steel frame the rest could also be laser cut from MDF, but personally I think it might be worth the money.

ragflyer, 2017-03-03:
I heated the subframe in the oven, fixed a U-profile on each side, rearranged the Y-axis bearings for less resistance, leveled the bed again at 90°C, fixed the Z-axis sensor and the offsets in the firmware.
Center area is fine now, only issue left is the cable of the heatbed sometimes getting stuck on Y-axis homing.
The money we saved could be put in a box for a new steel framed printer :)

closed 2017-03-03 by ragflyer

LukasFreeze, 2017-03-04:
(Virtual) box started! ;-)