heidelberg-makerspace/dai-makerspace-do-something #129: disable wiki spam filter

LukasFreeze, 2017-03-30:
There appears to be a spam filter in our wiki, preventing the addition of aliexpress links for example.

Since we already keep a tight control over which users get editing rights, disabling it altogether should be easier than adding exceptions now and then.

chombium, 2017-03-31:
I think it would be better if the spam filter stays active. We should only adjust the Spam filter's whitelist here https://wiki.heidelberg-makerspace.de/wiki/MediaWiki:Spam-whitelist

emka, 2017-04-02:
Added aliexpress.com to the whitelist as @chombium suggested. @LukasFreeze should have these permissions as well. Please test if that worked and also include the link that was blocked here if it still does not work, then I can test it myself.

LukasFreeze, 2017-04-02:
@PaulGoldschmidt, could you try your original link (without "kurzelinks.de" again?

LukasFreeze, 2017-05-03:
I confirmed that it works now, thanks!

closed 2017-05-03 by LukasFreeze