heidelberg-makerspace/dai-makerspace-do-something #130: TX Screw for Mounting on Metal-plate

topeban, 2017-03-30:
Information : Maybe this is the Wrong place  8-( 

Excess TX Screws used for Originally Mounting  
RITTAL  Cableholder for Flexible-Cable-Pipe to Metal-Plate in Electric-Cabinett
look here  e.g. :
http://www.rittal.com/de-de/product/show/variantdetail.action?productID=2589000

The Screw are equipment for 1 of 4 different Diameter (Can not Remember which)

http://www.rittal.com/de-de/product/list/variations.action?categoryPath=/PG0001/PG0900ZUBEHOER1/PG0918ZUBEHOER1/PG1174ZUBEHOER1/PGRP4073ZUBEHOER1/PRO13468ZUBEHOER1&productID=PRO13468

LukasFreeze, 2017-03-30:
I'm not sure what you mean - do you need screws or are you offering them?

Deutsch geht auch ;-)

topeban, 2017-04-03:
Na schön ,
Ich hab letzte Woche Schrauben in einer Schachtel vorbeigebracht (Abgepackt zu 20St.)
Linsenkopf mit TORX (25/30 o.ä.) für Befestigung von Kabelschlauchhaltern auf 
Stahlblech-Montageplatten.
Die Größe muss man ausmessen , bei Rittal steht nix von den (Beipack-)Schrauben.
Sonst wird da jeder Beipack-Mist dokumentiert in einer Liste.

Ich wollte nur dokumentieren wo die Schrauben herkommen bzw. Ihren Verwendungszweck.

mfg Thomas

closed 2017-04-03 by topeban

LukasFreeze, 2017-04-03:
Ah, cool, danke! Ich schau mir die Schachtel am Mittwoch mal an wenn ich
Zeit habe, ist sicher eine Schublade für frei :-)