heidelberg-makerspace/dai-makerspace-do-something #131: Wiki GIF "thumb" problem

PaulGoldschmidt, 2017-03-31:
Problem:
If you upload a GIF to the wiki and want to use it on a page with the "thoumb" attribute, you get the following error:

Error creating thumbnail: /srv/mediawiki/mediawiki-1.27.1/includes/limit.sh: line 101: 9528 Aborted /usr/bin/timeout $MW_WALL_CLOCK_LIMIT /bin/bash -c "$1" 3>&-

Error code: 134

chombium, 2017-03-31:
Exit code 134 means your program was aborted (received SIGABRT)... and the [assertion failed](https://www.student.cs.uwaterloo.ca/~cs136/seashell/docs/seashell-error-codes.html#%28part._.Error_code_134_-_.Program_.Abort%29)

I guess the allowed  memory for the shell commands is not enough to finish the thumb conversion.
@emka: try to raise the [$wgMaxShellMemory](https://www.mediawiki.org/wiki/Manual:$wgMaxShellMemory) or simply for test set it to 0 (no limit).

@PaulGoldschmidt: Please add a link to your GIF so that we can test

emka, 2017-04-02:
We had the default limits, I increased them a bit and the thumbnail has been generated successfully.

closed 2017-04-02 by emka

chombium, 2017-04-02:
cool. thanks @emka