heidelberg-makerspace/dai-makerspace-do-something #133: Setup a SSL certificate for the wiki with Letsencrypt

chombium, 2017-10-22:
It would be great if we secure our wiki with a SSL certificate.
One option would be to do that with [letsencrypt](https://letsencrypt.org)
It' easy and free to setup a certificate with letsencrypt

 Tutorials:

https://github.com/riceissa/nginx-mediawiki/blob/master/README.md
https://www.guyrutenberg.com/2015/11/07/getting-started-with-lets-encrypt-tutorial/
https://phschaad.com/guides/server_setup/ssl_with_letsencrypt

Best Regards,
Jovan

chombium, 2017-10-24:
@emka yes,  i was referring  to dai makerspace wiki

HDValentin, 2017-11-08:
Der Link von https://www.heidelberg-makerspace.de/ ist ohne https:// 
Hier wäre eine entsprechende Weiterleitung per .htaccess hilfreich.

Beispielcode:

RewriteEngine On
RewriteCond %{HTTPS} !=on
RewriteCond %{ENV:HTTPS} !=on
RewriteRule .* https://%{HTTP_HOST}%{REQUEST_URI} [R=307,L]
Header set Strict-Transport-Security "max-age=31536000" env=HTTPS