heidelberg-makerspace/dai-makerspace-do-something #134: correkt redirect to wiki

HDValentin, 2017-11-08:
I have just seen that the wiki has mooved or changed.
Is ist possible to redirect links that refer to the old wiki to the coresponding links in the new wiki?
For example:
https://wiki.heidelberg-makerspace.de/wiki/Project:Wallet

LukasFreeze, 2017-11-10:
Hi Valentin, we tried to make the process as smooth as possible with accompanying information in the community newsletter but of course that does not include everyone, so I'm sorry that the information didn't reach you earlier!

Are you referring to the now non-functional links on the DAI Makerspace facebook page? I agree those should be updated to say wiki.dai-makerspace... etc. but it appears some pages, including your example, have not been migrated yet. But I guess that's going to happen soon, Jasper should know the details :-)

HDValentin, 2017-11-10:
Hi Likas, thank you fpr the reply.  
> Are you referring to the now non-functional links on the DAI Makerspace facebook page?
Yes I think these people don't know about the wiki change at all.

devurandom, 2018-06-01:
https://www.heidelberg-makerspace.de/ links to http://wiki.dai-makerspace.de/ which does not redirect to https://wiki.comaking.space/

emka, 2018-06-02:
Originally it was planned to split the content specific to each location in August 2017, several causes lead to one of the two wikis still being almost empty.

Should we remove the link to wiki.dai-makerspace.de until that wiki is populated with more content?